tlambert03/nd2 #30: Symbols and libraries required in vendored binaries

jaimergp, 2021-11-12:
To make sure the binaries are conda-forge compatible, I did the following:

```
$> docker run -it condaforge/linux-anvil-cos7-x86_64 bash
# now inside docker:
$> mamba create -n nd2 -c conda-forge compilers sysroot_linux-64=2.17 python pip cython numpy libtiff libzlib jpeg 
$> conda activate nd2
$> git clone https://github.com/tlambert03/nd2
$> cd nd2

$> LD_LIBRARY_PATH=$CONDA_PREFIX/lib ldd -r -v src/sdk/Linux/lib/liblimfile.so 
src/sdk/Linux/lib/liblimfile.so: /opt/conda/envs/nd2/lib/libtiff.so.5: no version information available (required by src/sdk/Linux/lib/liblimfile.so)
        linux-vdso.so.1 =>  (0x00007fff97fb4000)
        libz.so.1 => /opt/conda/envs/nd2/lib/libz.so.1 (0x00007f16e597e000)
        libtiff.so.5 => /opt/conda/envs/nd2/lib/libtiff.so.5 (0x00007f16e58e9000)
        libpthread.so.0 => /lib64/libpthread.so.0 (0x00007f16e5190000)
        libstdc++.so.6 => /opt/conda/envs/nd2/lib/libstdc++.so.6 (0x00007f16e4fe5000)
        libm.so.6 => /lib64/libm.so.6 (0x00007f16e4ce3000)
        libgcc_s.so.1 => /opt/conda/envs/nd2/lib/libgcc_s.so.1 (0x00007f16e58d0000)
        libc.so.6 => /lib64/libc.so.6 (0x00007f16e4915000)
        libwebp.so.7 => /opt/conda/envs/nd2/lib/./libwebp.so.7 (0x00007f16e5847000)
        libzstd.so.1 => /opt/conda/envs/nd2/lib/./libzstd.so.1 (0x00007f16e4827000)
        liblzma.so.5 => /opt/conda/envs/nd2/lib/./liblzma.so.5 (0x00007f16e581e000)
        libLerc.so => /opt/conda/envs/nd2/lib/./libLerc.so (0x00007f16e478a000)
        libjpeg.so.9 => /opt/conda/envs/nd2/lib/./libjpeg.so.9 (0x00007f16e57df000)
        libdeflate.so.0 => /opt/conda/envs/nd2/lib/./libdeflate.so.0 (0x00007f16e57cf000)
        /lib64/ld-linux-x86-64.so.2 (0x00007f16e5778000)
        librt.so.1 => /lib64/librt.so.1 (0x00007f16e4582000)

        Version information:
        src/sdk/Linux/lib/liblimfile.so:
                libgcc_s.so.1 (GCC_3.0) => /opt/conda/envs/nd2/lib/libgcc_s.so.1
                libz.so.1 (ZLIB_1.2.0) => /opt/conda/envs/nd2/lib/libz.so.1
                libm.so.6 (GLIBC_2.2.5) => /lib64/libm.so.6
                libtiff.so.5 (LIBTIFF_4.0) => not found
                libpthread.so.0 (GLIBC_2.2.5) => /lib64/libpthread.so.0
                libc.so.6 (GLIBC_2.14) => /lib64/libc.so.6
                libc.so.6 (GLIBC_2.3.4) => /lib64/libc.so.6
                libc.so.6 (GLIBC_2.6) => /lib64/libc.so.6
                libc.so.6 (GLIBC_2.3) => /lib64/libc.so.6
                libc.so.6 (GLIBC_2.2.5) => /lib64/libc.so.6
                libc.so.6 (GLIBC_2.4) => /lib64/libc.so.6
                libstdc++.so.6 (GLIBCXX_3.4.20) => /opt/conda/envs/nd2/lib/libstdc++.so.6
                libstdc++.so.6 (CXXABI_1.3.8) => /opt/conda/envs/nd2/lib/libstdc++.so.6
                libstdc++.so.6 (GLIBCXX_3.4.17) => /opt/conda/envs/nd2/lib/libstdc++.so.6
                libstdc++.so.6 (CXXABI_1.3.9) => /opt/conda/envs/nd2/lib/libstdc++.so.6
                libstdc++.so.6 (GLIBCXX_3.4.9) => /opt/conda/envs/nd2/lib/libstdc++.so.6
                libstdc++.so.6 (CXXABI_1.3) => /opt/conda/envs/nd2/lib/libstdc++.so.6
                libstdc++.so.6 (GLIBCXX_3.4.14) => /opt/conda/envs/nd2/lib/libstdc++.so.6
                libstdc++.so.6 (GLIBCXX_3.4.11) => /opt/conda/envs/nd2/lib/libstdc++.so.6
                libstdc++.so.6 (CXXABI_1.3.5) => /opt/conda/envs/nd2/lib/libstdc++.so.6
                libstdc++.so.6 (GLIBCXX_3.4.15) => /opt/conda/envs/nd2/lib/libstdc++.so.6
                libstdc++.so.6 (GLIBCXX_3.4.22) => /opt/conda/envs/nd2/lib/libstdc++.so.6
                libstdc++.so.6 (GLIBCXX_3.4.21) => /opt/conda/envs/nd2/lib/libstdc++.so.6
                libstdc++.so.6 (GLIBCXX_3.4) => /opt/conda/envs/nd2/lib/libstdc++.so.6
        /opt/conda/envs/nd2/lib/libz.so.1:
                libc.so.6 (GLIBC_2.4) => /lib64/libc.so.6
                libc.so.6 (GLIBC_2.2.5) => /lib64/libc.so.6
        /opt/conda/envs/nd2/lib/libtiff.so.5:
                libm.so.6 (GLIBC_2.2.5) => /lib64/libm.so.6
                liblzma.so.5 (XZ_5.0) => /opt/conda/envs/nd2/lib/./liblzma.so.5
                libjpeg.so.9 (LIBJPEG_9.0) => /opt/conda/envs/nd2/lib/./libjpeg.so.9
                libc.so.6 (GLIBC_2.3) => /lib64/libc.so.6
                libc.so.6 (GLIBC_2.11) => /lib64/libc.so.6
                libc.so.6 (GLIBC_2.4) => /lib64/libc.so.6
                libc.so.6 (GLIBC_2.3.4) => /lib64/libc.so.6
                libc.so.6 (GLIBC_2.2.5) => /lib64/libc.so.6
        /lib64/libpthread.so.0:
                ld-linux-x86-64.so.2 (GLIBC_2.2.5) => /lib64/ld-linux-x86-64.so.2
                ld-linux-x86-64.so.2 (GLIBC_2.3) => /lib64/ld-linux-x86-64.so.2
[... 195 lines cut ...]
                libc.so.6 (GLIBC_2.2.5) => /lib64/libc.so.6
                libstdc++.so.6 (GLIBCXX_3.4.21) => /opt/conda/envs/nd2/lib/libstdc++.so.6
                libstdc++.so.6 (CXXABI_1.3.9) => /opt/conda/envs/nd2/lib/libstdc++.so.6
                libstdc++.so.6 (CXXABI_1.3) => /opt/conda/envs/nd2/lib/libstdc++.so.6
                libstdc++.so.6 (GLIBCXX_3.4) => /opt/conda/envs/nd2/lib/libstdc++.so.6
        /opt/conda/envs/nd2/lib/./libjpeg.so.9:
                libc.so.6 (GLIBC_2.3.4) => /lib64/libc.so.6
                libc.so.6 (GLIBC_2.7) => /lib64/libc.so.6
                libc.so.6 (GLIBC_2.4) => /lib64/libc.so.6
                libc.so.6 (GLIBC_2.2.5) => /lib64/libc.so.6
        /opt/conda/envs/nd2/lib/./libdeflate.so.0:
                libc.so.6 (GLIBC_2.3.4) => /lib64/libc.so.6
                libc.so.6 (GLIBC_2.4) => /lib64/libc.so.6
                libc.so.6 (GLIBC_2.2.5) => /lib64/libc.so.6
        /lib64/librt.so.1:
                libpthread.so.0 (GLIBC_2.3.2) => /lib64/libpthread.so.0
                libpthread.so.0 (GLIBC_PRIVATE) => /lib64/libpthread.so.0
                libpthread.so.0 (GLIBC_2.2.5) => /lib64/libpthread.so.0
                libc.so.6 (GLIBC_2.14) => /lib64/libc.so.6
                libc.so.6 (GLIBC_2.3.2) => /lib64/libc.so.6
                libc.so.6 (GLIBC_PRIVATE) => /lib64/libc.so.6
                libc.so.6 (GLIBC_2.2.5) => /lib64/libc.so.6
```

Looks like everything is in order. LD_LIBRARY_PATH is needed but this should be patched once `conda-build` does its magic.

jaimergp, 2021-11-12:
On OSX everything's static I think:

```
$> otool -L -v src/sdk/Darwin/lib/liblimfile.a
Archive : liblimfile.a
liblimfile.a(IoBaseDevice.cpp.o):
liblimfile.a(IoFileDevice.cpp.o):
liblimfile.a(IoFileDeviceImpl.cpp.o):
liblimfile.a(IoImageDataDevice.cpp.o):
liblimfile.a(IoMemoryBuffer.cpp.o):
liblimfile.a(IoMemoryBufferImpl.cpp.o):
liblimfile.a(IoMemoryDevice.cpp.o):
liblimfile.a(IoMemoryDeviceImpl.cpp.o):
liblimfile.a(StringConversions.cpp.o):
liblimfile.a(alloc.cpp.o):
liblimfile.a(Base64Coder.cpp.o):
liblimfile.a(xmlparse.c.o):
liblimfile.a(xmlrole.c.o):
liblimfile.a(xmltok.c.o):
liblimfile.a(CLxSerializerXMLString.cpp.o):
liblimfile.a(CLxStrBuffA.cpp.o):
liblimfile.a(CLxStrBuffW.cpp.o):
liblimfile.a(CLxStringA.cpp.o):
liblimfile.a(CLxStringW.cpp.o):
liblimfile.a(LxStringTokenizer.cpp.o):
liblimfile.a(LxStringUtils.cpp.o):
liblimfile.a(CLxByteArray.cpp.o):
liblimfile.a(CLxList.cpp.o):
liblimfile.a(CLxList_specI.cpp.o):
liblimfile.a(CLxVariant.cpp.o):
liblimfile.a(JsonBridge.cpp.o):
liblimfile.a(IoImageFile.cpp.o):
liblimfile.a(Nd2ChunkedDevice.cpp.o):
liblimfile.a(Nd2ChunkedDeviceImpl.cpp.o):
liblimfile.a(Nd2FileDevice.cpp.o):
liblimfile.a(Nd2FileDeviceImpl.cpp.o):
liblimfile.a(Nd2FileConversions.cpp.o):
liblimfile.a(TifFileDevice.cpp.o):
liblimfile.a(TifFileDeviceImpl.cpp.o):
liblimfile.a(TifLimFormat.cpp.o):
liblimfile.a(JsonFileDevice.cpp.o):
liblimfile.a(JsonFileDeviceImpl.cpp.o):

$> otool -L src/sdk/Darwin/lib/libnd2readsdk-static.a 
Archive : libnd2readsdk-static.a
libnd2readsdk-static.a(Nd2ReadSdk.cpp.o):
```

tlambert03, 2021-11-12:
wow, thanks so much for this!  really helpful to have that docker script.

closed 2021-11-15 by tlambert03