tlambert03/nd2 #38: nd2 not working on M1 architecture

zjniu, 2022-02-04:
* Python version: 3.10.2
* Operating System: macOS Big Sur 11.6

### Description

nd2 not working on M1 architecture.

### What I Did

I had tried running the package on a M1 iMac and ran into some issues with opening files. I downloaded the test data and tried to open a file as such.

```
import nd2

f = nd2.imread('nd2_test_data/aryeh_MeOh_high_fluo_007.nd2')
```

This generated the following error message.

```
Traceback (most recent call last):
  File "/Users/wniu/Documents/Code/test.py", line 3, in <module>
    f = nd2.imread('/Users/wniu/Downloads/nd2_test_data/aryeh_MeOh_high_fluo_007.nd2')
  File "/Users/wniu/miniforge3/envs/test/lib/python3.10/site-packages/nd2/nd2file.py", line 506, in imread
    with ND2File(file) as nd2:
  File "/Users/wniu/miniforge3/envs/test/lib/python3.10/site-packages/nd2/nd2file.py", line 54, in __init__
    self._rdr = get_reader(self._path)
  File "/Users/wniu/miniforge3/envs/test/lib/python3.10/site-packages/nd2/_util.py", line 24, in get_reader
    from ._sdk.latest import ND2Reader
  File "/Users/wniu/miniforge3/envs/test/lib/python3.10/site-packages/nd2/_sdk/__init__.py", line 1, in <module>
    from . import latest
ImportError: dlopen(/Users/wniu/miniforge3/envs/test/lib/python3.10/site-packages/nd2/_sdk/latest.cpython-310-darwin.so, 2): Symbol not found: _Lim_DestroyPicture
  Referenced from: /Users/wniu/miniforge3/envs/test/lib/python3.10/site-packages/nd2/_sdk/latest.cpython-310-darwin.so
  Expected in: flat namespace
 in /Users/wniu/miniforge3/envs/test/lib/python3.10/site-packages/nd2/_sdk/latest.cpython-310-darwin.so
```

tlambert03, 2022-02-04:
Thanks for letting me know!  I'm not surprised to hear it,  Nikon hasn't released ARM-compiled libraries for their SDK yet, so we'll have to wait on this one for a bit.

zjniu, 2022-02-04:
Ah that makes! Do you think it would be possible to run the SDK through Rosetta 2?

tlambert03, 2022-02-04:
it's a good question!  I'm not sure, I wish I had one to test on but I don't :)
maybe something [like this](https://medium.com/thinknum/how-to-install-python-under-rosetta-2-f98c0865e012) would work?

zjniu, 2022-02-04:
Very similar to your suggestion, I was able to create a Rosetta conda environment per the instructions given in https://github.com/conda-forge/miniforge/issues/165. Using the environment, I was able to get the package to work. It would be nice to get native M1 support in the future, but for now this works. Thanks for the help and the quick replies!

tlambert03, 2022-02-04:
very cool!  I pinged the folks at nd2sdk to see if they have any plans to release native libraries  anytime soon

zjniu, 2022-02-04:
Great, thanks for the awesome work on this package!

tlambert03, 2022-08-19:
@zjniu, I've built some wheels that I hope will work natively on m1, but I'm unable to test them.  

If you have a moment, I'd be grateful if you could let me know if these work for you.  Download the following link, unzip it, then pip install the `arm64` wheel corresponding to your python version (e.g. for python 3.9, pip install `nd2-0.1.dev1+g6dfd4e0-cp39-cp39-macosx_11_0_arm64`)

https://github.com/tlambert03/nd2/suites/7889065996/artifacts/335188596

cheers

tlambert03, 2022-08-19:
going to optimistically close this issue with #87. v0.4.0 (now on pypi) should support M1.  But please let me know if you find otherwise!

closed 2022-08-19 by tlambert03

tlambert03, 2022-08-25:
thanks for letting me know!