tlambert03/nd2 #45: Incompatibility with numpy<=1.21

zjniu, 2022-03-12:
* nd2 version: 0.2.1
* Python version: 3.10.2
* Operating System: macOS 11.4

### Description

The package seems to be currently incompatible with numpy<=1.21. On numpy==1.22.3, it works just fine, but downgrading to 1.21.5 for example gives the following error.

`ValueError: numpy.ndarray size changed, may indicate binary incompatibility. Expected 96 from C header, got 88 from PyObject`

This is problematic because numba currently only supports numpy<=1.21 and it is very commonly used in a lot of packages.

### What I Did

```
import nd2
nd2.imread('some.nd2')
```

tlambert03, 2022-03-12:
thanks, yeah I've seen that too, need to fix the wheel.  Please try the conda version for now `conda install -c conda-forge nd2`

zjniu, 2022-03-13:
The conda version works just as expected, thanks! By any chance, will the pip version be fixed when the next version is released?

tlambert03, 2022-03-13:
Yes I think I know what needs to be changed, and will do so before I release again to pypi

closed 2022-03-13 by tlambert03

tlambert03, 2022-03-14:
v0.2.2 released on pypi, and should work now.  thanks

PR #46 (2022-03-13): Fix numpy pinning in wheels
fixes #45