tlambert03/nd2 #49: Pip Installation fails - Double requirement given: numpy

SachidanandAlle, 2022-05-11:
* nd2 version: 0.2.2
* Python version: 3.8.10
* Operating System: Ubuntu 20.x

### Description

Can not install nd2 package using simple PIP

### What I Did
`pip install nd2[legacy]`

```
$pip install nd2[legacy]
Collecting nd2[legacy]
  Using cached nd2-0.2.2.tar.gz (5.4 MB)
  Installing build dependencies ... error
  ERROR: Command errored out with exit status 1:
   command: /usr/bin/python3 /usr/lib/python3/dist-packages/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-mbhnace4/overlay --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple -- setuptools wheel cython numpy 'numpy==1.14.5; python_version=='"'"'3.7'"'"'' 'numpy==1.17.3; python_version=='"'"'3.8'"'"'' 'numpy==1.19.3; python_version=='"'"'3.9'"'"'' 'numpy==1.21.3; python_version=='"'"'3.10'"'"''
       cwd: None
  Complete output (2 lines):
  Ignoring numpy: markers 'python_version == "3.7"' don't match your environment
  ERROR: Double requirement given: numpy==1.17.3 (already in numpy, name='numpy')
  ----------------------------------------
ERROR: Command errored out with exit status 1: /usr/bin/python3 /usr/lib/python3/dist-packages/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-mbhnace4/overlay --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple -- setuptools wheel cython numpy 'numpy==1.14.5; python_version=='"'"'3.7'"'"'' 'numpy==1.17.3; python_version=='"'"'3.8'"'"'' 'numpy==1.19.3; python_version=='"'"'3.9'"'"'' 'numpy==1.21.3; python_version=='"'"'3.10'"'"'' Check the logs for full command output.

$ pip list | grep numpy
numpy                       1.22.3              

```

tlambert03, 2022-05-11:
Ah interesting!  Thanks for reporting, until I have time to debug, please try a conda install for now (from the conda-forge channel)

SachidanandAlle, 2022-05-11:
Thanks for looking into this.. for now.. i could unblock by using `--no-build-isolation` but no idea of consequences.

tlambert03, 2022-05-11:
Also useful to know.  Thank you

aaristov, 2022-05-12:
Hi, I tried to reproduce the issue in Ubuntu and it just works. Could you make sure you're using the latest version of setuptools by running `pip install -U setuptools`?

tlambert03, 2022-05-26:
@SachidanandAlle, please also try `pip install -U pip` and let me know how it goes for you

tlambert03, 2022-06-06:
@SachidanandAlle, I have a feeling that if a pip upgrade didn't do it for you, then the last release v0.2.5 will do it.  I'm going to close this, but feel free to reopen if you can reproduce

closed 2022-06-06 by tlambert03

SachidanandAlle, 2022-06-06:
> pip install -U setuptools

This solves the problem. Thanks for looking into this