tlambert03/nd2 #72: ND2 files saved with lossless compression do not open correctly.

kthorn, 2022-06-23:
* nd2 version: 0.2.5
* Python version: 3.9.9
* Operating System: Windows 10 Pro

### Description

Files acquired by NIS Elements JOBS can be saved as ND2 files with either no compression or lossless compression. When saved with no compression, the nd2 library opens these files correctly. When saved with lossless compression, nd2 returns incorrect data.

### What I Did
I acquired two images of background (no sample) using JOBS, with and without saving with compression. You can see the results below - the version with no compression returns values close to 100 as expected for a background image; the version with compression returns nonsense values.
```
dir = r'C:\Users\kthorn\ArrePath, Inc\ArrePath Science - Documents\Imaging experiments\062322 Format Tests/'
with nd2.ND2File(dir+'NoCompression.nd2') as nd2_file:
    nd2_image = nd2_file.asarray()
    imdims = nd2_image.shape

nd2_image[1,0,0:10,0]
array([124, 108, 131, 113, 148, 131, 120, 127, 121, 129], dtype=uint16)

with nd2.ND2File(dir+'LosslessCompression.nd2') as nd2_file:
    nd2_image = nd2_file.asarray()
    imdims = nd2_image.shape

nd2_image[1,0,0:10,0]
with nd2.ND2File(dir+'LosslessCompression.nd2') as nd2_file:
    nd2_image = nd2_file.asarray()
    imdims = nd2_image.shape

nd2_image[1,0,0:10,0]
with nd2.ND2File(dir+'LosslessCompression.nd2') as nd2_file:
    nd2_image = nd2_file.asarray()
    imdims = nd2_image.shape
​
nd2_image[1,0,0:10,0]
array([40056, 11979, 13782, 37643,  2349, 20231,  1135, 16315, 57624,
       54800], dtype=uint16)
```

I can provide the example files that reproduce this problem.

tlambert03, 2022-06-23:
howdie @kthorn!  yeah some files would be great.  I suspect this will be pretty easy to fix

kthorn, 2022-06-23:
Want me to just send them to your email (they're <1MB)? I don't have a good place to host them.

tlambert03, 2022-06-23:
yep that works great.

closed 2022-06-25 by tlambert03

tlambert03, 2022-06-25:
should be fixed in main.  Just pushed a new release tag, should be built and on pypi in an hour or so

PR #74 (2022-06-25): feat: add `read_using_sdk` parameter, default to True for compressed files
fixes #72

This exposes the ability to use the sdk to read frames vs the current behavior of reading the chunkmap and using a numpy memmap.  This _may_ also help speed in reading metadata (#71)

The current default of `None` means: use the sdk reader if the file is compressed, otherwise use numpy memmap.
This default might change in the future