tlambert03/nd2 #76: nd2[legacy] dependency

c0deizbae, 2022-06-28:
Not sure if anyone else encounter this?

* nd2 version: latest
* Python version: 3.8.10
* Operating System: Ubuntu

### Description

When installing `nd2[legacy]` through `pip` outside of `condo` environment, wurlizter fails to install due to the tag `extra`.

### What I Did

```
pip install nd2[legacy]

ERROR: Could not find a version that satisfies the requirement wurlizter; extra == "legacy" (from nd2[legacy]) (from versions: none)
ERROR: No matching distribution found for wurlizter; extra == "legacy" (from nd2[legacy])
```

tlambert03, 2022-07-07:
ah shoot!  typo!  thanks

c0deizbae, 2022-07-08:
Didn't notice it was a typo. Submitted a PR.  #79

closed 2022-07-08 by tlambert03