tlambert03/nd2 #78: Position names are 'None'

gatoniel, 2022-07-07:
* nd2 version:  4.50.00
* Python version: 3.9.5
* Operating System: Windows 11 21H2 (Build 22000.739)

### Description

I would like to find out the names of the different positions I gave during the experiment. This should be the M-axis in NIS Elements.

### What I Did

I guess they should be in
```
nd2_file.experiment[1].parameters.points[i].name
```
Where `nd2_file.experiment[1]` is a `XYPosLoop`.

However, these values are of type `None`. In NIS Elements the different positions have names.

tlambert03, 2022-07-07:
Hey @gatoniel,  Thanks for the note.

yeah, seems like that would be the place.  lemme see what I can find.   Do you by chance have a smallish multipoint file where you _know_ what the position names are that you could send me?  (I can create one if not).  I just loaded a couple files in the actual nis viewer and also couldn't find position names, but I'm not sure if it's just those files...

gatoniel, 2022-07-07:
Thakns for the fast reply! @tlambert03 I just send a filesender link to you per e-mail

gatoniel, 2022-08-27:
Hey, did you have a chance to work out how to expose the position names to the API? If you do not have time for it, could you tell me the details on where to find the names in the nd2-file and where you wish to expose it in the API? I would try to add it myself in a PR.

tlambert03, 2022-08-27:
Hey @gatoniel, sorry for the delay.

I haven't yet exposed it, the code to decode the proper metadata chunk is in this branch: https://github.com/tlambert03/nd2/blob/decodemeta/src/nd2/_nd2decode.py

since i think this is ultimately a bug with the SDK too, I opened an issue with laboratory imaging; it would be ideal if this "just happened correctly" when using the sdk, and it'd be awesome if I didn't have to engineer some workaround for something they'll fix soon.  that said, the code isn't terribly far from usable.    I'm away this week, so it definitely won't happen for a couple weeks, but i haven't forgotten!

gatoniel, 2022-08-28:
Hey @tlambert03 , ok, if they are on it on the SDK side, I will just be patient and wait for it from that side. Thanks for the update.

tlambert03, 2022-09-11:
Hey @gatoniel, I just merged #93 which should fix missing names in `nd2_file.experiment[n].parameters.points[i].name` ... it also adds a new `unstructured_metadata()` method that exposes the full header metadata.  It's not carefully parsed or structured, but there's good stuff in there.  (for the position name, `ND2file.experiment` will automatically extract position name for you).  the 0.4.3 release is building now and it will be ready in 30-60 minutes or so

closed 2022-09-11 by tlambert03

gatoniel, 2022-09-12:
Just tried it out and it works perfectly well for me. Thanks!!!

PR #93 (2022-09-11): Add `unstructured_metadata` method, fix missing position names in experiment.
the new `unstructured_metadata` method exposes all metadata in the `ImageMetadataLV` portion of the nd2 header.  There's useful stuff in there.

this also closes #78 by mining that information for position names when the SDK fails to extract them (also contacted LIM about this)