tlambert03/nd2 #88: Some datasets appear skewed in Ubuntu

aaristov, 2022-08-31:
* nd2 version: 0.4.0
* Python version: 3.8.12
* Operating System: Ubuntu

### Description

Opening some large-scale (tiled) nd2 files in Ubuntu appears skewed. Same files read correctly in Windows.

### What I Did

I used AICSImageIO plugin in napari or nd2.ND2File().to_dask() - the result is the same as in the attachment. 
[image]

Ubuntu-report:
```bash
{
  "Version": "20.04",
  "OEM": {
    "Vendor": "Dell Inc.",
    "Product": "Precision 3630 Tower",
    "Family": "Precision",
    "DCD": "canonical-oem-somerville-bionic-amd64-20180608-47+beaver-turtlebay+X59"
  },
  "BIOS": {
    "Vendor": "Dell Inc.",
    "Version": "2.8.0"
  },
  "CPU": {
    "OpMode": "32-bit, 64-bit",
    "CPUs": "12",
    "Threads": "2",
    "Cores": "6",
    "Sockets": "1",
    "Vendor": "GenuineIntel",
    "Family": "6",
    "Model": "158",
    "Stepping": "10",
    "Name": "Intel(R) Core(TM) i7-8700 CPU @ 3.20GHz",
    "Virtualization": "VT-x"
  },
  "Arch": "amd64",
  "GPU": [
    {
      "Vendor": "10de",
      "Model": "1c30"
    }
  ],
  "RAM": 32.7,
  "Partitions": [
    485.3,
    0.8
  ],
  "Screens": [
    {
      "Size": "527mmx296mm",
      "Resolution": "1920x1080",
      "Frequency": "60.00"
    }
  ],
  "Autologin": false,
  "LivePatch": false,
  "Session": {
    "DE": "ubuntu:GNOME",
    "Name": "ubuntu",
    "Type": "x11"
  },
  "Language": "en_US",
  "Timezone": "Europe/Paris",
  "Install": {
    "Media": "Ubuntu 18.04 \"Bionic\" - Build amd64 LIVE Binary 20180608-09:38\n",
    "Type": "GTK",
    "DownloadUpdates": true,
    "Language": "en",
    "Minimal": false,
    "RestrictedAddons": false,
    "Stages": {
      "0": "language",
      "3": "user_done",
      "271": "done"
    }
  }
}
```

pip freeze:
```bash
aicsimageio==4.9.2
aicspylibczi==3.0.5
aiohttp==3.8.1
aiosignal==1.2.0
alabaster==0.7.12
apoc==0.7.0
appdirs==1.4.4
argon2-cffi==21.3.0
argon2-cffi-bindings==21.2.0
asciitree==0.3.3
astropy==5.0.2
asttokens==2.0.5
async-timeout==4.0.2
attrs==21.4.0
autopep8==1.6.0
Babel==2.9.1
backcall==0.2.0
bbii-decon==0.0.1
beautifulsoup4==4.11.1
bioformats-jar==6.7.0.post2
black==22.1.0
bleach==5.0.0
cachey==0.2.1
cellpose==2.0.5
cellpose-napari==0.1.4
certifi==2021.10.8
cffi==1.15.0
charset-normalizer==2.0.12
click==8.0.4
cloudpickle==2.0.0
cupy==10.2.0
cycler==0.11.0
dask==2022.2.0
debugpy==1.5.1
decorator==5.1.1
defusedxml==0.7.1
dnspython==2.2.1
docstring-parser==0.13
docutils==0.17.1
elementpath==2.5.0
email-validator==1.1.3
entrypoints==0.4
executing==0.8.2
fasteners==0.17.3
fastjsonschema==2.15.3
fastremap==1.12.2
fastrlock==0.8
fire==0.4.0
fonttools==4.33.3
freetype-py==2.2.0
frozenlist==1.3.0
fsspec==2022.2.0
HeapDict==1.0.1
hsluv==5.0.2
idna==3.3
imagecodecs==2022.2.22
imageio==2.10.5
imageio-ffmpeg==0.4.5
imagesize==1.3.0
importlib-metadata==4.11.1
importlib-resources==5.4.0
intervaltree==3.1.0
ipykernel==6.9.1
ipython==8.0.1
ipython-genutils==0.2.0
ipywidgets==7.7.0
jedi==0.18.1
Jinja2==3.0.3
joblib==1.1.0
JPype1==1.3.0
jsonschema==4.4.0
jupyter==1.0.0
jupyter-client==7.1.2
jupyter-console==6.4.3
jupyter-core==4.9.2
jupyterlab-pygments==0.2.2
jupyterlab-widgets==1.1.0
jupytext==1.13.8
kiwisolver==1.3.2
llvmlite==0.38.0
locket==0.2.1
loguru==0.6.0
lxml==4.8.0
magicgui==0.3.7
markdown-it-py==2.1.0
MarkupSafe==2.1.0
matplotlib==3.5.2
matplotlib-inline==0.1.3
mdit-py-plugins==0.3.0
mdurl==0.1.1
mistune==0.8.4
mrc==0.2.0
multidict==6.0.2
mypy-extensions==0.4.3
napari==0.4.15
napari-accelerated-pixel-and-object-classification==0.7.3
napari-aicsimageio==0.6.1
napari-assistant==0.2.0
napari-console==0.0.4
napari-cupy-image-processing==0.2.3
napari-elementary-numpy-operations==0.0.5
napari-plugin-engine==0.2.0
napari-pyclesperanto-assistant==0.18.0
-e git+https://github.com/aaristov/napari-segment.git@aef1dbb101ba39bd0880c62a8aba1947064dcfa7#egg=napari_segment
napari-skimage-regionprops==0.5.0
napari-stracking==0.1.9
napari-svg==0.1.6
napari-time-slicer==0.4.3
napari-tools-menu==0.1.11
napari-workflows==0.1.8
natsort==8.1.0
nbclient==0.6.2
nbconvert==6.5.0
nbformat==5.4.0
nd2==0.4.0
nest-asyncio==1.5.4
networkx==2.6.3
notebook==6.4.11
npe2==0.1.2
numba==0.55.1
numcodecs==0.9.1
numpy==1.21.6
numpydoc==1.2
ome-types==0.2.10
opencv-python-headless==4.5.5.64
packaging==21.3
pandas==1.4.1
pandocfilters==1.5.0
parso==0.8.3
partd==1.2.0
pathspec==0.9.0
pexpect==4.8.0
pickleshare==0.7.5
Pillow==8.4.0
Pint==0.18
platformdirs==2.5.1
pooch==1.6.0
prometheus-client==0.14.1
prompt-toolkit==3.0.28
psutil==5.9.0
psygnal==0.3.3
ptyprocess==0.7.0
pure-eval==0.2.2
pyclesperanto-prototype==0.17.1
pycodestyle==2.8.0
pycparser==2.21
pydantic==1.9.0
pyerfa==2.0.0.1
Pygments==2.11.2
pyopencl==2022.1.3
PyOpenGL==3.1.6
pyparsing==3.0.7
pyperclip==1.8.2
pypher==0.7.0
PyQt5==5.15.6
PyQt5-Qt5==5.15.2
PyQt5-sip==12.9.1
pyrsistent==0.18.1
python-dateutil==2.8.2
pytomlpp==1.0.10
pytools==2022.1.6
pytz==2021.3
PyWavelets==1.2.0
PyYAML==6.0
pyzmq==22.3.0
qtconsole==5.2.2
QtPy==2.0.1
readlif==0.6.5
requests==2.27.1
resource-backed-dask-array==0.1.0
scikit-image==0.19.2
scikit-learn==1.0.2
scipy==1.8.0
Send2Trash==1.8.0
setuptools-scm==6.4.2
six==1.16.0
snowballstemmer==2.2.0
sortedcontainers==2.4.0
soupsieve==2.3.2.post1
Sphinx==4.4.0
sphinxcontrib-applehelp==1.0.2
sphinxcontrib-devhelp==1.0.2
sphinxcontrib-htmlhelp==2.0.0
sphinxcontrib-jsmath==1.0.1
sphinxcontrib-qthelp==1.0.3
sphinxcontrib-serializinghtml==1.1.5
stack-data==0.2.0
stracking==0.1.9
superqt==0.3.0
termcolor==1.1.0
terminado==0.13.3
threadpoolctl==3.1.0
tifffile==2022.2.9
tinycss2==1.1.1
toml==0.10.2
tomli==2.0.1
toolz==0.11.2
torch==1.11.0
tornado==6.1
tqdm==4.62.3
traitlets==5.1.1
transforms3d==0.3.1
typer==0.4.0
typing_extensions==4.1.1
urllib3==1.26.8
vispy==0.9.6
wcwidth==0.2.5
webencodings==0.5.1
widgetsnbextension==3.6.0
wrapt==1.13.3
wurlitzer==3.0.2
xarray==2022.3.0
xmlschema==1.9.2
yarl==1.7.2
zarr==2.11.0
-e git+ssh://git@gitlab.pasteur.fr/aaristov/zarr-tools.git@3ab710131768d8be8b6af88111916f646ce2153f#egg=zarr_tools
zipp==3.7.0
```

aaristov, 2022-08-31:
This is the link to the original dataset https://github.com/BaroudLab/anchor-droplet-chip/releases/download/v0.0.5-pre/BF.nd2

tlambert03, 2022-08-31:
nevermind, was also able to reproduce on mac, and it looks like that flag isn't making a difference

PR #90 (2022-09-08): fix: Fix images where widthPx x Bytes is not the same as expected width Bytes
fixes #88  (which was also showing in a couple datasets).

Occasionally, `attributes.widthPx` will not be the same as `attributes.widthBytes // (attributes.componentCount * attributes.bitsPerComponentInMemory // 8)`, as was the case with @aaristov's dataset in #88.

In those cases _both_ the sdk reader and the native memmap reader result in skewed images (see #88).  Oddly enough, the NIS elements viewer is fine with these files, suggesting they don't directly use the SDK reader internally :/

This PR catches those ca [... cut]
closed 2022-09-08 by tlambert03