tlambert03/nd2 #105: cannot find metadata

DanBushey, 2022-10-27:
* nd2 version: 0.4.6
* Python version: 3.10.6
* Operating System: Ubuntu 22.04

### Description
I am searching for metadata but cannot find it.
I know the metadata exists because when using NIS elements software I can export as an excel file:
https://hhmionline-my.sharepoint.com/:x:/g/personal/busheyd_hhmi_org/EdGxdrwO_o9Ml1m3M6NS7WsBxXaAv2Ej9y7Ra-dwyFPR-g?e=aycWd2
I am specifically looking for data in sheet "Recorded Data" column 640 nm[%].

The nd2 file can be found here (large file 18.6 GB):
https://hhmionline-my.sharepoint.com/:u:/g/personal/busheyd_hhmi_org/ESt_jvLHeMVIhiXF3VAk4aYB0NZYxLe3HT4OVH6T8s_iZA?e=zwHEbt

### What I Did
I've looked in (f= nd2.ND2File(target_file)):
f.frame_metadata(0) #looking at first frame. Assuming the 640 nm would be here per frame in time series image.
Also, I looked in other places but do  not indicate per frame state:
f.custom_data - can see 640 nm as tag 6
f.attributes
f.experiment
f.unstructured_metadata() #when running this it hangs
f.metadata

Assistance is greatly appreciated. Although I can export the data via elements it takes a long time.
Thanks for nd2

tlambert03, 2022-10-28:
thanks for the detailed report and data.  being addressed in #107

tlambert03, 2022-10-29:
after the next release (coming today) you'll be able to access this at `f.recorded_data`.  It will be a dict of numpy arrays or sequences, and is suitable for passing to pandas dataframe.

```python
f = nd2.ND2File('some_file.nd2)
df = pd.DataFrame(f.recorded_data)
```

closed 2022-10-29 by tlambert03

DanBushey, 2022-10-30:
Thank you so much. With the upgrade to 0.5.0 it works great. Also, pulling the timestamp from recorded_data is much faster.

PR #107 (2022-10-29): feat: Extract recorded data
closes #105 

@DanBushey ... this will mostly resolve your issue in #105.  I'm still unsatisfied with the extraction of strings (like the DMD column in your data) ... but all the other columns are coming out ok.  Will get this in soon, thanks for the report!