tlambert03/nd2 #110: please pull from main... the download script was recently updated:

winston1716, 2022-11-01:
please pull from main... the download script was recently updated:

https://github.com/tlambert03/nd2/blob/ec0f00e5e4af2f9528f85e5b8ae0fc293e65ad96/scripts/download_samples.py#L9

_Originally posted by @tlambert03 in https://github.com/tlambert03/nd2/issues/103#issuecomment-1273517560_

winston1716, 2022-11-01:
The updated URL also returns to a 404 page, could you update it again? Thank you.

tlambert03, 2022-11-01:
May i ask why you opened the issue?

It's true that the URL for our internal testing data may change periodically.  But the only "goal" is that the link works for our tests here.  It's not intended as an external resource.

closed 2022-11-01 by tlambert03

winston1716, 2022-11-01:
Sorry, I mistakenly thought this was a public resource, so I applied for access.

tlambert03, 2022-11-01:
In any case, follow the instructions in your first post "please pull from main" and you'll see the new link

Point being, the url is in a script that nd2 uses for testing.  If your url goes stale, it means that your repo is out of date and you need to update your repository from main.

Note that the url is already changed in https://github.com/tlambert03/nd2/blob/main/scripts/download_samples.py

winston1716, 2022-11-02:
Thank you for your answer, I checked the scripts again and my problem has been resolved. 

I've just started to use GitHub recently and there are some community rules that I'm not very familiar with. I will pay more attention, thank you for your reminding.

Wish you all the best.