tlambert03/nd2 #114: File handle left open when an nd2 file fails to read

manthey, 2022-12-07:
* nd2 version: 0.5.1
* Python version: 3.9
* Operating System: Ubuntu 20.04

### Description

I try to open the nd2 file downloaded from here: https://downloads.openmicroscopy.org/images/ND2/aryeh/b16_pdtB+y50__crop.nd2
It fails with `TypeError: <lambda>() missing 6 required positional arguments: 'bitsPerComponentInMemory', 'bitsPerComponentSignificant', 'componentCount', 'heightPx', 'pixelDataType', and 'sequenceCount'`.

If I catch that exception and do something else, I notice that there is an open file handle to the file I tried to load that never closes, even when all my local variables are deleted and `gc.collect()` is run.  If I try to open a number of bad files, eventually, a lot of file handles are permanently open (cab be checked via /proc/<pid>/fd, for instance).

### What I Did

```
python test.py

# where test.py is
---
import gc
import time
import nd2

try:
    f = nd2.ND2File("b16_pdtB+y50__crop.nd2")
    print(f.sizes)
except Exception as exc:
    print exc
f = None
gc.collect()
time.sleep(1000)
---

ls /proc/<pid of python process>/fd
```
Naturally, it would be great to read all files, but on failure it would be nice to release the file handle.  I tried explicitly calling f.close() and f._rdr.close() in the exception handler, but that didn't release it either.

tlambert03, 2022-12-07:
just a note the `b16_pdtB` file in your link has the same sha as my local one, and worked fine for me too... (`f.sizes = {'T': 57, 'P': 12, 'Y': 1200, 'X': 1600}`) will seek out a linux box to try as well

tlambert03, 2022-12-08:
If I catch that exception and do something else, I notice that there is an open file handle to the file I tried to load that never closes, even when all my local variables are deleted and gc.collect() is run.  If I try to open a number of bad files, eventually, a lot of file handles are permanently open (cab be checked via /proc//fd, for instance).

manthey, 2022-12-08:
Thanks @tlambert03.  I still see the problem on Ubuntu.  Specifically, I think the file handle is staying open after whatever causes the failure in reading the attributes within the file there.  I also just tried this on python 3.9.15 on an M1 mac, and there it works (and tells me I failed to close the handle before garbage collection).

In ordinary use, I am using `close()`.  That doesn't affect whether the file handle stays open on Ubuntu, though.  I also tried explicitly calling `f._rdr.close()` in the exception handler, but that didn't release it either.

And, I doubt it will be surprising, but the failure also occurs on Ubuntu using python 3.11.

manthey, 2022-12-08:
As a crude workaround for the file handle, if I guard these lines: https://github.com/tlambert03/nd2/blob/main/src/nd2/_sdk/latest.pyx#L78-L84 in a try/except, and then on exception do `Lim_FileClose(self._fh)` and `self._is_open = 0` and then reraise, it does release the file handle.

tlambert03, 2022-12-08:
thanks, yeah, I definitely can/will do that too... 

but I really want to figure out why you're getting that lambda error! :joy:  I just tried on my ubuntu (20.04.5, py311) and the file opened fine! 
I'm having difficulty even determining where that error would happen (I don't have any lambdas in the code).  do you have a full stack trace?

tlambert03, 2022-12-08:
interesting.  I'm wondering if `Lim_FileGetAttributes` in the SDK itself is somehow failing on those systems.

if it's not too much trouble, could you test that for me by putting the following lines immediately after the opening of `self._fh` (around line 78 in latest.pyx)?

```python
            string = Lim_FileGetAttributes(self._fh)
            print(string)
            Lim_FileFreeString(string)
```

manthey, 2022-12-08:
Hmm... That `print(string)` causes a Seg fault.

Based on this working in a couple of different docker environments, I'm guessing it is somehow a library I have on my Ubuntu machines that is causing an issue.

tlambert03, 2022-12-08:
nonetheless ... concerning!  Do _any_ files open for you on those systems?  or nothing at all?

tlambert03, 2022-12-08:
and it's the `print` statement huh... not the `Lim_FileGetAttributes(self._fh)` call?

manthey, 2022-12-08:
Other nd2 files from that same directory work: (e.g., MeOh_high_fluo_003.nd2), plus lots I have from another lab work correctly.

I'm not aware of using that json library,

closed 2022-12-08 by tlambert03

tlambert03, 2022-12-08:
very interesting! thanks so much.  I'm going to open a new issue to track this linux error

PR #117 (2022-12-08): fix: catch exception on file open and release handle
fixes #114 

@manthey ... this look good?