tlambert03/nd2 #118: bug: Lim_FileGetAttributes failing in certain linux environments

tlambert03, 2022-12-08:
As one more interesting clue to failing to read some nd2 files in some environments and not others: When I use stock Ubuntu python 3.8 or python 3.9 installed from the deadsnakes ppa, I get the failure to read attributes.  If I use pyenv to compile python 3.9 locally, that reads the nd2 file successfully.  The only difference I have noticed is that using LD_TRACE_LOADED_OBJECTS=1 shows that the system/deadsnakes pythons both pull in libexpat, whereas the pyenv python does not.

_Originally posted by @manthey in https://github.com/tlambert03/nd2/issues/114#issuecomment-1343355668_

manthey, 2022-12-09:
Here is a way to reproduce this with docker:

This Dockerfile works:
```
FROM python:3.9-slim

COPY b16_pdtB+y50__crop.nd2 .

RUN python3 -m pip install virtualenv
RUN python3 -m virtualenv /venv
RUN /venv/bin/pip install nd2

RUN /venv/bin/python -c 'import nd2;print(nd2.ND2File("b16_pdtB+y50__crop.nd2").sizes)'
```
The output is:
```
<string>:1: UserWarning: ND2File file not closed before garbage collection. Please use `with ND2File(...):` context or call `.close()`.
{'T': 57, 'P': 12, 'Y': 1200, 'X': 1600}
```

This Dockerfile fails:
```
FROM ubuntu:20.04

RUN apt-get update && apt-get install -y python3-pip

COPY b16_pdtB+y50__crop.nd2 .

RUN python3 -m pip install virtualenv
RUN python3 -m virtualenv /venv
RUN /venv/bin/pip install nd2

RUN /venv/bin/python -c 'import nd2;print(nd2.ND2File("b16_pdtB+y50__crop.nd2").sizes)'
```
The output is:
```
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/venv/lib/python3.8/site-packages/nd2/nd2file.py", line 98, in __init__
    self._rdr = get_reader(
  File "/venv/lib/python3.8/site-packages/nd2/_util.py", line 68, in get_reader
    return ND2Reader(
  File "src/nd2/_sdk/latest.pyx", line 48, in nd2._sdk.latest.ND2Reader.__cinit__
  File "src/nd2/_sdk/latest.pyx", line 79, in nd2._sdk.latest.ND2Reader.open
  File "src/nd2/_sdk/latest.pyx", line 115, in nd2._sdk.latest.ND2Reader.attributes.__get__
TypeError: __new__() missing 6 required positional arguments: 'bitsPerComponentInMemory', 'bitsPerComponentSignificant', 'componentCount', 'heightPx', 'pixelDataType', and 'sequenceCount'
```

tlambert03, 2023-06-14:
with #135, we now no longer depend on any C code (libexpat is irrelevant now).  So I'm going to optimistically close this

closed 2023-06-14 by tlambert03

manthey, 2023-06-14:
Excellent.