tlambert03/nd2 #126: Package not found on conda-forge channels

tyjsamo, 2023-02-28:
* nd2 version:
* Python version: 3.11.0
* Operating System: macOS

### Description
I'd like to install nd2 via conda-forge, but I'm getting a PackagesNotFoundError. 

### What I Did
Attempted to install nd2 into environment directly during environment creation and after creation using install command.
```
conda create -n py-nd2 -c conda-forge python=3.11 nd2
Collecting package metadata (current_repodata.json): done
Solving environment: failed with repodata from current_repodata.json, will retry with next repodata source.
Collecting package metadata (repodata.json): done
Solving environment: failed

PackagesNotFoundError: The following packages are not available from current channels:

  - nd2

Current channels:

  - https://conda.anaconda.org/conda-forge/osx-arm64
  - https://conda.anaconda.org/conda-forge/noarch
  - https://repo.anaconda.com/pkgs/main/osx-arm64
  - https://repo.anaconda.com/pkgs/main/noarch
  - https://repo.anaconda.com/pkgs/r/osx-arm64
  - https://repo.anaconda.com/pkgs/r/noarch

To search for alternate channels that may provide the conda package you're
looking for, navigate to

    https://anaconda.org

and use the search bar at the top of the page.

conda install -n py-nd2 -c conda-forge nd2
Collecting package metadata (current_repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Collecting package metadata (repodata.json): /done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.

PackagesNotFoundError: The following packages are not available from current channels:

  - nd2

Current channels:

  - https://conda.anaconda.org/conda-forge/osx-arm64
  - https://conda.anaconda.org/conda-forge/noarch
  - https://repo.anaconda.com/pkgs/main/osx-arm64
  - https://repo.anaconda.com/pkgs/main/noarch
  - https://repo.anaconda.com/pkgs/r/osx-arm64
  - https://repo.anaconda.com/pkgs/r/noarch

To search for alternate channels that may provide the conda package you're
looking for, navigate to

    https://anaconda.org

and use the search bar at the top of the page.
```

tlambert03, 2023-02-28:
i assume you're on an m1...

can you try the `https://conda.anaconda.org/conda-forge/osx-64` channel?  due to the way this package is bundled (the binaries are compiled outside of the conda build process) it's conceivable that the macos package there *will* work for you, even though it's not in the osx-arm64 channel

metalmicrobe, 2023-02-28:
Thanks! This worked. Correct, I'm using an m1. 

It possibly goes without saying, but in case other m1 users see this and are planning to use nd2 with other packages (or an IDE such as spyder, as I am), you'll need to create a new environment that specifically uses the https://conda.anaconda.org/conda-forge/osx-64 channel to install all packages and ensure there aren't errors due to accessing the default arm64 files that `conda install -n myenv -c conda-forge 'package-name'` will automatically install.

tlambert03, 2023-02-28:
Thanks for confirming, I'll try to get this fixed soon as well

closed 2023-06-15 by tlambert03