tlambert03/nd2 #172: Getting ND2 series images in the order they were captured

mendelfishman, 2023-09-15:
* nd2 version: 0.5.3
* Python version: 3.114
* Operating System: Windows 10

### Description

Using a multipoint image file and the metadata on the stage position for each field of view in the series, I am trying to stitch the field of view together into a single image. The problem I have encountered is the series is loaded into a third dimension of an array in what seems to be an incorrect order. Is there something I am missing or a workaround to this problem?

Thank you!

### What I Did

```
# Get all image data
f = nd2.ND2File(image_path)
raw_image_array = img_as_float(f)

# Get metadata on the x,y, and z positions

# Populate the dictionary with integers as keys from 0 to 36, image_array and positional data as values

# Create an empty dictionary
xy_positions_dict = {}

# Populate the dictionary with integers as keys from 0 to 36, image_array and positional data as values
for i in range(fov_count):

    # Correct for abnormal layering of FOV series
    k = 1%6 + int(i/6)

    # Get tile array
    tile_array = raw_image_array[k,:,:]

    # Get full FOV string
    main_pattern = r'Position\([^)]+\)'
    full_string = re.findall(main_pattern, xy_positions_str)[i]

    # Get stage position
    stage_position = full_string[full_string.find("[")+len("["):full_string.find("]")]
    stage_position = [float(x) for x in stage_position.split(",")]

    # Get PFS offset
    pfs_off = full_string[full_string.find("pfsOffset=")+len("pfsOffset="):full_string.find(", name")]

    xy_positions_dict[i] = {'Image array': tile_array, 'x': stage_position[0], 'y': stage_position[1], 'z': stage_position[2], 'PFS offset': pfs_off}

# Define stitching function
def stitch_image(dict):
    
    # Create an empty array
    x_values_min = min([xy_positions_dict[i]['x'] for i in range(fov_count)])
    x_values_max = max([xy_positions_dict[i]['x'] for i in range(fov_count)])
    y_values_min = min([xy_positions_dict[i]['y'] for i in range(fov_count)])
    y_values_max = max([xy_positions_dict[i]['y'] for i in range(fov_count)])

    # Create a new array
    array_x_size = int((x_values_max - x_values_min) * vx_per_um[0]) + fov_x_length
    array_y_size = int((y_values_max - y_values_min) * vx_per_um[1]) + fov_y_length
    output_array = np.zeros((array_x_size, array_y_size), dtype=np.float64)
    output_array = output_array + 0
        
     for key in dict:

        # Get the image x and y values
        x = dict[key]['x']
        y = dict[key]['y']

        # Calculate the boundaries of the array
        x_l = int(vx_per_um[0] * (x - x_values_min))
        x_r = x_l + fov_x_length
        y_t = int(vx_per_um[1] * (y - y_values_min))
        y_b = y_t + fov_y_length
        # print(x_l, x_r, y_t, y_b)
        output_array[x_l:x_r, y_t:y_b] = dict[key]['Image array']

    return output_array

# Excecute image stitching function
unprocessed_output_array = stitch_image(xy_positions_dict)
```

tlambert03, 2023-09-15:
hey @mendelfishman, thanks for opening an issue.

I need a bit more information on what `img_as_float` is doing and where you're getting `fov_count`, and what `xy_positions_str` is...

is the image too large to share?  I can definitely try to help you achieve whatever it is you want to do, but I need a runnable example, or a bit more explicit information on where you're accessing the the metadata from the nd2 API, what you're expecting to see, and what you're actually seeing

mendelfishman, 2023-09-15:
Hi @tlambert03

I did some more digging and realized that the issue was specific to this .nd2 file, which NIS viewer also displays in the incorrect issue. Using other images, ND2 opens 4-dimensional images (multipoint z-stacks) properly.

Thank you.

tlambert03, 2023-09-15:
Thanks for the update!

closed 2023-09-15 by tlambert03