tlambert03/nd2 #173: order of dimensions

acycliq, 2023-09-15:
Apologies, this is not really an issue but is there a way to set the order of axes on the object that is returned from `nd2.ND2File`. For example when you read an nd2 file then the object has the dimensions if the format `TZCYX`. Can I set this order tp `TCZYX` please?

tlambert03, 2023-09-15:
the axes will be ordered, by default, as they were acquired/specified in the NDAquisition tab.  But you can easily transpose them:

1. using the named dimensions in the xarray output:
```python
import nd2

ary = nd2.imread('file.nd2', xarray=True).transpose('T', 'C', 'Z', ...)
```
2. Or, if you don't want to use xarray, you can do it manually 
```python
ary = nd2.imread('file.nd2').transpose(0,2,1,3,4)
```
... and if you're not certain that `C` is the second axis and `Z` is the first axis, then you can use `sizes` to know for sure:
```python
with ND2File('file.nd2') as f:
    ary = f.asarray()
    order = tuple(list(f.sizes).index(i) for i in 'TCZYX')
    ary = ary.transpose(order)
```

acycliq, 2023-09-15:
Excellent! Thanks. Doing `nd2.imread(img_path, dask=True).transpose(0,2,1,3,4)` works fine. For some reason however it crashes for me when I try `nd2.imread(img_path, dask=True).transpose('T','C','Z','Y','X')`. I guess it is a long shot but the last few lines of the call stack are:

```
File ~/miniconda3/envs/decon-napari/lib/python3.9/site-packages/dask/array/routines.py:181, in <genexpr>(.0)
    179     if len(axes) != a.ndim:
    180         raise ValueError("axes don't match array")
--> 181     axes = tuple(d + a.ndim if d < 0 else d for d in axes)
    182 else:
    183     axes = tuple(range(a.ndim))[::-1]

TypeError: '<' not supported between instances of 'str' and 'int'
```

Have I got something wrong please?

tlambert03, 2023-09-15:
using strings to transpose will only work if you use `xarray=True`, as in my first example (*not* `dask=True`, as you have there).  both dask and numpy expect integers to be passed to transpose... only the xarray output recognizes axes with string names

tlambert03, 2023-09-15:
if you also want to use dask, you can with `imread(..., xarray=True, dask=True)`, which will give you xarray that can be indexed with strings, but will also be backed by a delayed dask array

acycliq, 2023-09-15:
Thanks so much! That helps a lot. Re: the 'P', i need to check with the ppl who did the experiment I guess.

closed 2023-09-15 by acycliq

tlambert03, 2023-09-15:
sure. let me know if i can help you find metadata.  the "main" metadata describing the dimensions that the user acquired will be held in [ND2File.experiment](https://www.talleylambert.com/nd2/API/nd2/#nd2.ND2File.experiment) ... the axes are summarized in the [`ND2File.sizes`](https://www.talleylambert.com/nd2/API/nd2/#nd2.ND2File.sizes) dict (it's an ordered dict, so they will also be in the same order as the array output).

acycliq, 2023-09-15:
Thanks again! I think I should be able to handle the metadata