tlambert03/nd2 #182: Coordinates for RoiShapeType.Polygon not showing

pawlowska, 2023-10-10:
* nd2 version: NIS Elements 5.42.03 build 1812
* Operating System: Windows
### Description

I tried to export ROI coordinates for different types of ROI. 

### What I Did

Opened nd2 file. Drew two ROI, rectangle and polygon. Saved file. ROI types and centres show correctly, but only the rectangle has all coordinates. The polygon has five (0,0) instead of five corners.

```
with nd2.ND2File("histo.nd2") as f:
    print([(f.rois[k].info.label, f.rois[k].info.shapeType) for k in f.rois.keys()])
    print('')
    print(f.rois[1].animParams)
    print('')
    print(f.rois[2].animParams)
```

Output:
```
[('', <RoiShapeType.Rectangle: 3>), ('', <RoiShapeType.Polygon: 5>)]

[AnimParam(timeMs=0.0, enabled=True, centerX=-0.934640522875817, centerY=-0.9155555555555556, centerZ=0.0, rotationZ=0.0, boxShape=BoxShape(sizeX=0.126984126984127, sizeY=0.16444444444444442, sizeZ=0.0), extrudedShape=ExtrudedShape(sizeZ=0, basePoints=[]))]

[AnimParam(timeMs=0.0, enabled=True, centerX=-0.4860168501038321, centerY=-0.8736948690902574, centerZ=0.0, rotationZ=0.0, boxShape=BoxShape(sizeX=0.0, sizeY=0.0, sizeZ=0.0), extrudedShape=ExtrudedShape(sizeZ=0, basePoints=[XYPoint(x=0, y=0), XYPoint(x=0, y=0), XYPoint(x=0, y=0), XYPoint(x=0, y=0), XYPoint(x=0, y=0)]))]
```

tlambert03, 2023-10-10:
thanks for the report @pawlowska!  I will try to reproduce it here soon, but if you happen to have the nd2 file handy and could share it that would be very helpful.

tlambert03, 2023-10-10:
never mind... was able to create a file that reproduced it.  will look into it

tlambert03, 2023-10-10:
fixed in #183 and released in v0.8.0 which is building now and should appear on pypi shortly.  thanks for the report!

PR #183 (2023-10-10): fix: fix extraction of xy points in extruded shapes
fixes #182