tlambert03/nd2 #232: How to convert ROI coordinates from relative to pixels

Kees-van-der-Oord-Nikon, 2024-06-20:
* nd2 version: 5.42.06
* Python version: 3.9
* Operating System: W10

### Description

This is not really an issue, just a suggestion to add to the documentation

### What I Did

The code below converts the ROI coordinates from relative fractions to pixel coordinates.
The Ellipse ROI width and height are before rotation over the indicated angle.
The only missing detail is the inner circle of the Ring ROI.

```
f = nd2.ND2File(sys.argv[1])

mid_x = f.shape[1] / 2.
mid_y = f.shape[0] / 2.

def RelToPixSizeX(x):
	return mid_x * x

def RelToPixSizeY(y):
	return mid_y * y

def RelToPixPosX(x):
	return mid_x + mid_x * x

def RelToPixPosY(y):
	return mid_y + mid_y * y

for id in f.rois:
	roi = f.rois[id]
	header = str(roi.id) + ";" + nd2.structures.RoiShapeType(roi.info.shapeType).name + ";"
	pnts = ""
	info = roi.animParams[0]
	if (roi.info.shapeType == nd2.structures.RoiShapeType.Rectangle) or \
			(roi.info.shapeType == nd2.structures.RoiShapeType.Square) or \
			(roi.info.shapeType == nd2.structures.RoiShapeType.Circle) or \
			(roi.info.shapeType == nd2.structures.RoiShapeType.Ellipse) or \
			(roi.info.shapeType == nd2.structures.RoiShapeType.Ring):
		pnts = "(" + \
			"{:#.1f}".format(RelToPixPosX(info.centerX)) + "," + \
			"{:#.1f}".format(RelToPixPosY(info.centerY)) + "),(" + \
			"{:#.1f}".format(RelToPixSizeX(info.boxShape.sizeX)) + "," + \
			"{:#.1f}".format(RelToPixSizeY(info.boxShape.sizeY)) + ")"
	elif (roi.info.shapeType == nd2.structures.RoiShapeType.Line) or \
			(roi.info.shapeType == nd2.structures.RoiShapeType.Polygon) or \
			(roi.info.shapeType == nd2.structures.RoiShapeType.PolyLine) or \
			(roi.info.shapeType == nd2.structures.RoiShapeType.Bezier):
		for pnt in roi.animParams[0].extrudedShape.basePoints:
			if len(pnts) != 0:
				pnts += ","
			pnts += "(" + "{:#.1f}".format(RelToPixPosX(pnt.x+info.centerX)) + "," + "{:#.1f}".format(RelToPixPosY(pnt.y+info.centerY)) + ")"
	print(header + pnts + ";" + "{:#.4f}".format(info.rotationZ))

f.close()
```
WR,

Kees

tlambert03, 2024-06-20:
hey @Kees-van-der-Oord-Nikon

Thanks for your note.  I agree, it's pretty cumbersome to use ROIs straight from the file metadata :joy:
This is great and i'd be happy to add a convenience like this to the code base.  Would you be interested in opening a PR to that effect?  (Happy to do it myself if not)

Kees-van-der-Oord-Nikon, 2024-06-21:
Dear Talley,

Thanks for your response.
I think you are more fluent in github than me, so please do the PR.
The biggest challenge was that f.shape array has the dimensions in an unexpected order (height,width).

I think the interface should support moving ROIs as well.
I expect that the animsParam array has more entries in that case (one for every control frame ?)

WR,

Kees

Dr. Kees van der Oord
European Support Specialist
T: +31207099246
M: +31623367589
E: ***@***.***
Website
Nikon Europe B.V.
Stroombaan 14
1181 VX Amstelveen
Netherlands

Registered Chamber of Commerce Amsterdam No. 34139593
VAT No. NL 8091.99.889.B.01

This e-mail may contain information that is PRIVILEGED and/or CONFIDENTIAL. This e-mail is intended solely for the individual or entity to whom it is addressed. If you are not, or have reason to believe you are not the intended recipient of this e-mail please contact us by reply e-mail as soon as possible. Immediately after contacting us, please destroy this e-mail (including attachment(s)) from your computer system(s). Any unauthorized copying, disclosure or distribution of this e-mail (including attachment(s)) is not allowed. Nikon Europe BV hereby disclaims any and all liabilities which may arise in relation to any of its electronic communication. No obligation is entered into by virtue of this e-mail, unless confirmed in writing and duly signed by an authorized representative of Nikon Europe BV

tlambert03, 2024-06-21:
> please do the PR.

no worries, will do

> The biggest challenge was that f.shape array has the dimensions in an unexpected order (height,width).

yes (height, width) (row major) is the standard convention for arrays in python, numpy, etc.., so `nd2` sticks with this convention.  it's true that needs to be taken into consideration when comparing to the file metadata